autowarefoundation/autoware_core #142: Port ' autoware_gnss_poser ' and 'autoware_universe_utils' from Autoware Universe to Autoware Core

liuXinGangChina, 2024-12-31:
### Checklist

- [X] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [X] I've searched other issues and no duplicate issues were found.
- [X] I've agreed with the maintainers that I can plan this task.

### Description

As part of Autoware Core implementation activity discussed in this [thread](https://github.com/orgs/autowarefoundation/discussions/5365), I will be porting autoware_gnss_poser from Autoware Universe to Autoware Core.

### Purpose

Port autoware_gnss_poser  from Autoware Universe to Autoware Core to make autowre-core functional

### Possible approaches

Port dependent packages from Universe to Core.

### Definition of done

- [ ]  port autoware_test_utils

liuXinGangChina, 2025-01-02:
1. except for some msg depend on package "autoware_core_component_interface_specs", which will be solved by this [pr](https://github.com/autowarefoundation/autoware.core/pull/124), node can pass complie process with no error
2. i will upload a draft pr for this node "autoware_gnss_poser "

xmfcx, 2025-02-19:
- https://github.com/autowarefoundation/autoware.universe/pull/9909
- https://github.com/autowarefoundation/autoware.core/pull/166

were merged, closing this issue. Thanks everyone!

closed 2025-02-19 by xmfcx