autowarefoundation/autoware_core #175: Port 'autoware_ekf_localizer' from Autoware Universe to Autoware Core

Motsu-san, 2025-01-20:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

As part of Autoware Core implementation activity discussed in this [thread](https://github.com/orgs/autowarefoundation/discussions/5365), I will be porting autoware_ekf_localizer from Autoware Universe to Autoware Core.

### Purpose

Port autoware_ekf_localizer from Autoware Universe to Autoware Core to make autowre-core functional

### Possible approaches

Port dependent packages from Universe to Core.

### Definition of done

- [x]  Refactor Readme file according to migration guidance
- [x]  Create draft pr
- [ ]  Pr merged

Motsu-san, 2025-02-04:
Missing autoware_universe_utils porting, These PRs were reverted. So I'm in progress again.

YamatoAndo, 2025-04-11:
@Motsu-san If this task has been completed, please close the issue.

closed 2025-04-18 by Motsu-san