autowarefoundation/autoware_core #198: Porting autoware_test_utils from autoware.universe to autoware.core

NorahXiong, 2025-02-17:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

As part of Autoware Core implementation activity discussed in this [thread](https://github.com/orgs/autowarefoundation/discussions/5365), I will be porting autoware_test_utils from Autoware Universe to Autoware Core.

### Purpose

Port autoware_test_utils from Autoware Universe to Autoware Core.

### Possible approaches

Fix bugs if detected. Deal dependency problems related to  autoware_test_utils if detected.

### Definition of done

- [ ] port autoware_test_utils
  - [ ] add autoware_test_utils to autoware.core according to conventions
  - [ ] remove autoware_test_utils from autoware.universe and fix dependency problem if detected

liuXinGangChina, 2025-02-18:
Good Morning Mits san and Fatih san @mitsudome-r @xmfcx 

Shall we put port this message under the autoware.core/testing folder since it is a test related library package?

Best regards

心刚

xmfcx, 2025-02-18:
Good morning @liuXinGangChina 🌅

Yes, I've checked the contents of `autoware_planning_test_manager` and it can be moved into this new `autoware.core/testing` sub-folder ✅

cc. @takayuki5168 @soblin @maxime-clem