autowarefoundation/autoware_core #226: Porting autoware_motion_velocity_obstacle_stop_module from autoware.universe to autoware.core

NorahXiong, 2025-02-26:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

As part of Autoware Core implementation activity discussed in this [thread](https://github.com/orgs/autowarefoundation/discussions/5365), I will be porting autoware_test_utils from Autoware Universe to Autoware Core.

### Purpose

Port autoware_motion_velocity_obstacle_stop_module from Autoware Universe to Autoware Core.

### Possible approaches

Fix bugs if detected. Deal dependency problems related to autoware_motion_velocity_obstacle_stop_module if detected.

### Definition of done

- [ ] port autoware_motion_velocity_obstacle_stop_module

NorahXiong, 2025-02-27:
[image]

NorahXiong, 2025-02-27:
I will work on these dependencies is no one else assigned. cc @mitsudome-r @xmfcx @youtalk @liuXinGangChina  
- add referenced msg from tier4_autoware_msgs to autoware_internal_msgs (tier4_metric_msgs)
- deal with dependence on autoware_planning_factor_interface (autoware.universe/planning)

liuXinGangChina, 2025-02-27:
Good Moring Mits san

 I look into the dependency graph，find that if we follow the porting rule in this [file ](https://docs.google.com/document/d/1kKshojZo8RuYQ-WhIqQCqTKQO3ojTb5RB77V3NjYpdU/edit?tab=t.0#heading=h.185kwwpluhpx)， we should move autoware_planning_factor_interface  first， but i didn‘t find it in the porting task [list](https://docs.google.com/spreadsheets/d/1B7w8AHo7MEDexLrvd-8BKDzORzPVg5P2w67g-_K83Vo/edit?pli=1&gid=0#gid=0)。

Can you check with the planning team whether this package “autoware_planning_factor_interface” is in the porting plan

Best regards

心刚

liuXinGangChina, 2025-02-28:
> Hello Mits san [@mitsudome-r](https://github.com/mitsudome-r)
> 
> I look into the dependency graph，find that if we follow the porting rule in this [file ](https://docs.google.com/document/d/1kKshojZo8RuYQ-WhIqQCqTKQO3ojTb5RB77V3NjYpdU/edit?tab=t.0#heading=h.185kwwpluhpx)， we should move autoware_planning_factor_interface first， but i didn‘t find it in the porting task [list](https://docs.google.com/spreadsheets/d/1B7w8AHo7MEDexLrvd-8BKDzORzPVg5P2w67g-_K83Vo/edit?pli=1&gid=0#gid=0)。
> 
> Can you check with the planning team whether this package “autoware_planning_factor_interface” is in the porting plan
> 
> If it is， we will port **autoware_planning_factor_interface** first
> 
> At the mean time， we are working on porting the msg first
> 
> Best regards
> 
> 心刚

Already got confirmation from mits san, go go go! @NorahXiong

NorahXiong, 2025-02-28:
> > Hello Mits san
> > I look into the dependency graph，find that if we follow the porting rule in this [file ](https://docs.google.com/document/d/1kKshojZo8RuYQ-WhIqQCqTKQO3ojTb5RB77V3NjYpdU/edit?tab=t.0#heading=h.185kwwpluhpx)， we should move autoware_planning_factor_interface first， but i didn‘t find it in the porting task [list](https://docs.google.com/spreadsheets/d/1B7w8AHo7MEDexLrvd-8BKDzORzPVg5P2w67g-_K83Vo/edit?pli=1&gid=0#gid=0)。
> > Can you check with the planning team whether this package “autoware_planning_factor_interface” is in the porting plan
> > If it is， we will port **autoware_planning_factor_interface** first
> > At the mean time， we are working on porting the msg first
> > Best regards
> > 心刚
> 
> Already got confirmation from mits san, go go go! [@NorahXiong](https://github.com/NorahXiong)

Already done. https://github.com/autowarefoundation/autoware_internal_msgs/pull/52

liuXinGangChina, 2025-03-03:
> > > Hello Mits san
> > > I look into the dependency graph，find that if we follow the porting rule in this [file ](https://docs.google.com/document/d/1kKshojZo8RuYQ-WhIqQCqTKQO3ojTb5RB77V3NjYpdU/edit?tab=t.0#heading=h.185kwwpluhpx)， we should move autoware_planning_factor_interface first， but i didn‘t find it in the porting task [list](https://docs.google.com/spreadsheets/d/1B7w8AHo7MEDexLrvd-8BKDzORzPVg5P2w67g-_K83Vo/edit?pli=1&gid=0#gid=0)。
> > > Can you check with the planning team whether this package “autoware_planning_factor_interface” is in the porting plan
> > > If it is， we will port **autoware_planning_factor_interface** first
> > > At the mean time， we are working on porting the msg first
> > > Best regards
> > > 心刚
> > 
> > 
> > Already got confirmation from mits san, go go go! [@NorahXiong](https://github.com/NorahXiong)
> 
> Already done. [autowarefoundation/autoware_internal_msgs#52](https://github.com/autowarefoundation/autoware_internal_msgs/pull/52) At this moment, porting is blocked by [autoware_planning_factor_interface](https://github.com/autowarefoundation/autoware.core/issues/226#issuecomment-2686719075).

https://github.com/autowarefoundation/autoware.universe/pull/10204 autoware_planning_factor_interface waiting for review

closed 2025-03-27 by mitsudome-r