autowarefoundation/autoware_core #266: implement crop_box_filter

mitsudome-r, 2025-03-12:


liuXinGangChina, 2025-03-12:
Will creat a plan on how to implement  for sensing team review first

liuXinGangChina, 2025-03-13:
# background #
Implement a crop-box node to filter input pointcloud，without dependency to universe::pointcloudpreprocessor::filter base class

# reference code #
The code is going to be created based on current corp_box_filter_node code
* https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/autoware_pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_node.cpp
* https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/autoware_pointcloud_preprocessor/include/autoware/pointcloud_preprocessor/crop_box_filter/crop_box_filter_node.hpp
* https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/autoware_pointcloud_preprocessor/include/autoware/pointcloud_preprocessor/filter.hpp
* https://github.com/autowarefoundation/autoware.universe/blob/main/sensing/autoware_pointcloud_preprocessor/src/filter.cpp

# Code Structure After refactor # 

## construction function ##
* get parameters from config file
* init subscriber for inout pointcloud 
* init publisher for polygon and filtered cloud
* init debug tools like stop watch

## input pointcloud callback ##
* verify the input pointcloud's field -- referenced from https://github.com/autowarefoundation/autoware.universe/blob/bcea43e045af23c53e61681749d7d61d7ab0533d/sensing/autoware_pointcloud_preprocessor/src/filter.cpp#L368
* filter the pointcloud
* construct output pointcloud
* publish filtered pointcloud and polygon
* publish debug info if needed

p.s. 
* use scoped lock to ensure memory safety
* use memcpy to speed up processing time

## need discussed ##
There are dynamic parameters re-configure mechanism in original cropbox code
But actually in most scenarios( >99%), we ignore that, because in a real-time project, point cloud crop parameters are calculated at very beginning stage and stay the same for the reset of the time.

So, what is your idea about the re-configure mechanism, shall i implement it so that we can keep the crop-box-filter-node its original taste? @mitsudome-r  @xmfcx 

Best regards

心刚

xmfcx, 2025-03-13:
Thanks for coming up with a comprehensive implementation plan 💡

Reconfigure is useful for setting up the boundaries when you set up the vehicle, is as useful as the polygon for initial setup. I think it'd bee good to have, we already have the implementation [here](https://github.com/autowarefoundation/autoware.universe/blob/bcea43e045af23c53e61681749d7d61d7ab0533d/sensing/autoware_pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_node.cpp#L263) anyways.

Maybe we could add an improvement over the original code too:
- https://github.com/autowarefoundation/autoware.universe/blob/bcea43e045af23c53e61681749d7d61d7ab0533d/sensing/autoware_pointcloud_preprocessor/src/crop_box_filter/crop_box_filter_node.cpp#L191

Here it's building up and publishing this debug polygon even if no one is using it. And it's not even controlled by some debug parameter, some waste of resources.
It could have a conditional on it like:
```cpp
if (crop_box_polygon_pub_->get_subscription_count() > 0) {
    publishCropBoxPolygon();
}
```

liuXinGangChina, 2025-03-14:
Morning Fatih san @xmfcx 

Thanks for your quick reply

1. I will keep the original parameters reconfigure method
2. will add a checker for polygon pulisher

Have a nice day

心刚

liuXinGangChina, 2025-03-14:
Code completed, testing now

will be released tomorrow if everything gose well

心刚

closed 2025-03-25 by mitsudome-r