autowarefoundation/autoware_core #288: Mismatched namespace of LoggerLevelConfigure

NorahXiong, 2025-03-19:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I'm convinced that this is not my fault but a bug.

### Description

Link error occurred in the unit test after splitting the [autoware_utils_logging](https://github.com/autowarefoundation/autoware_utils/pull/37) package.

It could be fixed by modifying the namespace and dependence but the build will fail as the [build_depends.repos](https://github.com/autowarefoundation/autoware.core/pull/239) is abandoned while the split packages have not been deployed in the container yet.

### Expected behavior

All tests can pass.

### Actual behavior

Test action failed due to link errors.

### Steps to reproduce

1. add test of autoware_ekf_localizer
2. trigger tests in github CICD

NorahXiong, 2025-03-19:
@youtalk Need help.

NorahXiong, 2025-03-19:
> [@NorahXiong](https://github.com/NorahXiong) Is there a link to failing CI?

https://github.com/autowarefoundation/autoware_core/actions/runs/13941474517/job/39019002722?pr=289

youtalk, 2025-03-21:
I think it will be resolved by bumping up `autoware_utils` to 1.3.0.
https://github.com/autowarefoundation/autoware_utils/

I will do it.

youtalk, 2025-03-24:
The latest `autoware` images have `autoware_utils` 1.3.0. https://github.com/autowarefoundation/autoware/pull/5906

@NorahXiong You need to include `autoware_utils_logging` instead of `autoware_utils/logging` if you want to use `autoware_utils` 1.3.0.

closed 2025-03-24 by youtalk