autowarefoundation/autoware_core #403: Replace autoware_utils in control component

isamu-takagi, 2025-04-16:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

Replace `autoware_utils` with the split `autoware_utils_*` package.

### Purpose

Remove unnecessary dependencies.

### Possible approaches

Refer to https://github.com/autowarefoundation/autoware_utils/issues/34 for the replacement package.

### Definition of done

- [ ] autoware_core/control/autoware_simple_pure_pursuit

a-maumau, 2025-04-18:
working in progress.

a-maumau, 2025-04-18:
PR: https://github.com/autowarefoundation/autoware_core/pull/416