autowarefoundation/autoware_core #434: Reimplement downsample filters

mitsudome-r, 2025-04-23:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

We should port downsample filters to process pointclouds before ndt_scan_matcher and euclidean cluster.

For this we want to port downsample modules from [pointcloud_preprocessor](https://autowarefoundation.github.io/autoware_universe/main/sensing/autoware_pointcloud_preprocessor/docs/downsample-filter/).

We would like to port it as a new package with the following modification:
* create a new package called autoware_downsample_filters
* remove dependency on Filter base class (just like we did for crop_box_filter)
* Our first target can be [Voxel Grid Downsample Filter](https://autowarefoundation.github.io/autoware_universe/main/sensing/autoware_pointcloud_preprocessor/docs/downsample-filter/#voxel-grid-downsample-filter)
  * As extended goal, we can port [Pickup Based Voxel Grid Downsample Filter](https://autowarefoundation.github.io/autoware_universe/main/sensing/autoware_pointcloud_preprocessor/docs/downsample-filter/#pickup-based-voxel-grid-downsample-filter) and [Random Downsample Filter](https://autowarefoundation.github.io/autoware_universe/main/sensing/autoware_pointcloud_preprocessor/docs/downsample-filter/#random-downsample-filter)

### Purpose

Explained in description

### Possible approaches

Explained in Description

### Definition of done

- [ ] re-implement Voxel Grid Downsample Filter with a newly created pacakge
- [ ] add Pickup Based Voxel Grid Downsample Filter
- [ ] add Random Downsample Filter

NorahXiong, 2025-04-23:
# Source code comparison (L - PCL, R - AW)

[source code in PCL](http://github.com/PointCloudLibrary/pcl/blob/master/apps/cloud_composer/tools/voxel_grid_downsample.cpp#L44-L52)
[source code in Autoware](https://github.com/autowarefoundation/autoware_universe/blob/main/sensing/autoware_pointcloud_preprocessor/src/downsample_filter/voxel_grid_downsample_filter_node.cpp#L92-L96)

[image]

# Similarities
Both use PCL's VoxelGrid filter as the core algorithm
Both allow configuration of voxel sizes in x, y and z dimensions
Both preserve the original point cloud metadata
# Differences
**Implementation context:**
- Autoware: Real-time ROS 2 component for robotics/autonomous driving
- PCL: Interactive GUI tool for point cloud processing

**Performance considerations:**
- Autoware: Has a "faster_filter" implementation for performance optimization
- PCL: Simpler implementation focused on interactive use

**Execution model:**
- Autoware: Designed for streaming data processing
- PCL: Single-execution tool with GUI interaction

**Parameter handling:**
- Autoware: Dynamic parameter updates via ROS 2 parameter system
- PCL: Parameters exposed through Qt property model for GUI

**Thread safety:**
- Autoware: Uses mutex locking for thread safety
- PCL: No explicit thread-safety mechanisms

liuXinGangChina, 2025-04-24:
> # Voxel Grid Downsample Filter comparison (L - PCL, R - AW)
> 
> [image] [source code in PCL](https://github.com/PointCloudLibrary/pcl/blob/master/apps/cloud_composer/tools/voxel_grid_downsample.cpp#L44-L52) [source code in Autoware](https://github.com/autowarefoundation/autoware_universe/blob/main/sensing/autoware_pointcloud_preprocessor/src/downsample_filter/voxel_grid_downsample_filter_node.cpp#L92-L96)
> # Similarities
> 
> Both use PCL's VoxelGrid filter as the core algorithm Both allow configuration of voxel sizes in x, y and z dimensions Both preserve the original point cloud metadata
> # Differences
> 
> **Implementation context:**
> 
>     * Autoware: Real-time ROS 2 component for robotics/autonomous driving
> 
>     * PCL: Interactive GUI tool for point cloud processing
> 
> 
> **Performance considerations:**
> 
>     * Autoware: Has a "faster_filter" implementation for performance optimization
> 
>     * PCL: Simpler implementation focused on interactive use
> 
> 
> **Execution model:**
> 
>     * Autoware: Designed for streaming data processing
> 
>     * PCL: Single-execution tool with GUI interaction
> 
> 
> **Parameter handling:**
> 
>     * Autoware: Dynamic parameter updates via ROS 2 parameter system
> 
>     * PCL: Parameters exposed through Qt property model for GUI
> 
> 
> **Thread safety:**
> 
>     * Autoware: Uses mutex locking for thread safety
> 
>     * PCL: No explicit thread-safety mechanisms

Thanks for the comparison Norah

From my point of view

I think we can remove the pcl official BSD license， since most of the pointcloud related developer using vexel_grid filter have to do like this ，and it truly different from original pcl tutorial code

What is your idea ， Kondo san @youtalk 

Best regards

心刚

closed 2025-05-28 by mitsudome-r