autowarefoundation/autoware_core #453: porting eculidean_cluster_node to autoware.core

liuXinGangChina, 2025-04-27:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

I am porting port Euclidean cluster to autoware core

following below process

1. port autoware-Euclidean-cluster to core and rename it to autoware_euclidean_cluster_core
2. replace output topic msg type from `tier4_perception_msgs::msg::DetectedObjectsWithFeature` to  `autoware_perception_msgs::msg::DetectedObjects`
3. test with awsim

### Purpose

Explained in description

### Possible approaches

Explained in description

### Definition of done

- [ ] core side pr created

liuXinGangChina, 2025-04-28:
@mitsudome-r  mits san

which solution do you think is better？

mitsudome-r, 2025-04-28:
For euclidean cluster, we can keep the one in autoware universe as is. (Just like we did for crop_box_filter)
We can create it as a new package. Ideally, it would be nice if we can name it autoware_euclidean_cluster in Autoware Core and rename the one in universe as autoware_euclidean_cluster_universe

mitsudome-r, 2025-04-30:
@liuXinGangChina I had a feedback from Perception team that they would like to keep the name in Autoware Universe. Instead, we can name the one in core as `autoware_euclidean_cluster_object_detector` package.

closed 2025-05-21 by mitsudome-r