autowarefoundation/autoware_core #454: porting ndt_scan_matcher to autoware.core

liuXinGangChina, 2025-04-28:
### Checklist

- [x] I've read the [contribution guidelines](https://github.com/autowarefoundation/autoware/blob/main/CONTRIBUTING.md).
- [x] I've searched other issues and no duplicate issues were found.
- [x] I've agreed with the maintainers that I can plan this task.

### Description

1. I am porting port ndt_scan_matcher to autoware core

there are 2 approches for that:

first solution
1. port ndt_scan_matcher to core and rename it to autoware_ndt_scan_matcher_core
2. replace service type from tier4_localization_msgs/srv/pose_with_covariance_stamped to autoware_internal_localization_msgs/srv/pose_with_covariance_stamped
3. test with awsim

second solution
1. replace service type from tier4_localization_msgs/srv/pose_with_covariance_stamped to autoware_internal_localization_msgs/srv/
2. resolve conflict caused by replacing service type above
3. port ndt_scan_matcher to core
4. remove port ndt_scan_matcher from universe side

### Purpose

Explained in description

### Possible approaches

Explained in description

### Definition of done

core side pr created

liuXinGangChina, 2025-04-28:
@mitsudome-r  mits san

which solution do you think is better？

mitsudome-r, 2025-04-28:
Let's move on with approach 2. We can fix the message type in Universe first and port it to core.

mitsudome-r, 2025-05-23:
@NorahXiong Sorry, I didn't notice the message from Lucas and did it myself so that we can initiate the discussion around licence terms. Thank you for your effort.